tomato64/tomato64 #16: Status > Device List page is slow to load

HommeOursPorc, 2023-09-28:
The Status > Device List page is slow to load. About 5 seconds. On FreshTomato it loads in a fraction of a second, it's almost instantaneous.

lancethepants, 2023-10-05:
The 5 second issue was a dnsmasq patch not properly being enabled.
https://github.com/tomato64/tomato64/commit/f1e9f0888ac56ff637afa228cb3abea81a474375

It's a lot quicker but it still takes ~1.5sec to load for some reason. There's still leftover wifi code that needs to commented out I'll have to take a closer look at.

lancethepants, 2023-10-05:
The lease times on the page now work too.

lancethepants, 2023-10-06:
https://github.com/tomato64/tomato64/commit/0a3de2736fac7987882cb915de3ec2d5819df09e

I did some digging and I think we were getting a timing issue on page loads.  Upon loading the Device List page tomato asynchronously tells dnsmasq to write all lease information to a file in memory and then checks if it's done before proceeding.  Well I think on faster hardware the check happens before dnsmasq can finish, so then we were stuck waiting a whole second before checking again.  I changed it to check every microsecond and now the page loads instantly!

I also removed the unneeded wifi columns.

lancethepants, 2023-10-06:
Should be available to test after tonight's nightly build.

HommeOursPorc, 2023-10-08:
I confirm the page now load fast :)

closed 2023-10-08 by HommeOursPorc