tomato64/tomato64 #17: Administration > Admin Access - Page theme doesn't repect selected theme.

HommeOursPorc, 2023-10-15:
This is my current theme:
[image]

When I open Admin Access:
[image]

Theme change. Ctrl + F5 doesn't change anything.

HommeOursPorc, 2023-10-15:
EDIT:

Theme is bugged. After saving some configs the theme switched for every pages... ???

[image]

HommeOursPorc, 2023-10-15:
Yup I think I'm the problem here and I didn't reset my browser's cache properly. All pages seems to be set to the Tomato theme now.

HommeOursPorc, 2023-10-15:
Theme is definitely bugged! Theme keep changing. Ctrl + F5 and Ctrl Shift R on Firefox doesn't change anything. The themes have a mind of their own ahah!

HommeOursPorc, 2023-10-16:
I need to test this again. After sleeping on it I think I got things mixed up and it's probably a user error... My error.

lancethepants, 2023-10-16:
I haven't seen this issue, but I install from scratch a bunch, so maybe if upgrading I couldn't maybe image this. I renamed the `default` them to `USB Black` I believe, but once you hit save on that page it should resolve any issues.

HommeOursPorc, 2023-10-16:
I upgraded from an older release with the .tzst file. What I think happened is that I needed to clear my cache but I didn't and that's why it was erratic. After, I can't remember for sure, but I think I changed it back to USB black but was checking some other things and when I finally saved the page I didn't remember that I've changed it. So the theme changed again, but again I didn't clear my cache so it could have been erratic again.

In the end, it's probably all my fault lol. I'll double check to confirm when I get a chance :)

HommeOursPorc, 2023-10-16:
Tested with an other browser and all works fine. My error, sorry about that.

closed 2023-10-16 by HommeOursPorc