tomato64/tomato64 #19: Administration > Admin Access - Page doesn't redirect correctly to selected WAN port when saving config changes

HommeOursPorc, 2023-10-15:
Administration > Admin Access - Page doesn't redirect correctly to selected WAN port when saving config changes. 
To be fair, this is probably a FreshTomato issue.

I currently use port 8080 for WAN access to the GUI. On the Admin Access page, when you save your config, the GUI seems to be redirecting me to the LAN port 80. I think I noticed this on some other pages too, but I currently can only confirm this behaviour for the Admin Access page.

When on the Admin Access page, notice the port 8080:
[image]

After the saving timer is done, the page timeout and I lose access to the GUI since it seems it's trying to get LAN port 80 instead:
[image]

lancethepants, 2023-11-06:
It looks to me that FreshTomato has the save behavior as well.

HommeOursPorc, 2024-05-09:
Fixed in FT? 
https://bitbucket.org/pedro311/freshtomato-arm/commits/84ab99bc53e1f5b5bac30c16d7d4f67a619fc117

lancethepants, 2024-05-09:
I was just about to post here. Yes, I believe so. Commits are already in Tomato64 too and available in the latest nightly.

closed 2024-05-15 by lancethepants