tomato64/tomato64 #21: Preserve Bandwidth Monitoring

lancethepants, 2023-11-06:


wellloaded, 2023-12-01:
I would also suggest to allow for a way longer period of time to be retained. If I remember well rstat stores up to 2 years worth of data which can be easily extended on modern ARM router & especially tomato64

HommeOursPorc, 2024-03-18:
> I would also suggest to allow for a way longer period of time to be retained. If I remember well rstat stores up to 2 years worth of data which can be easily extended on modern ARM router & especially tomato64

Kind of off topic, but I already did some test about this : 
https://www.linksysinfo.org/index.php?threads/is-there-a-way-to-configure-to-keep-wan-monthly-bandwidth-beyond-2-years.77412/

I finally never tried if it worked since I needed to start a new because I couldn't import my old data back in.

wellloaded, 2025-12-03:
Thanks! This opens the door to many possibilities. My fav one is a bandwidth usage heatmap in a matrix style e.g.
- per hour of the day
- period of the week (morning/afternoon/evening/night)
- weekday of the month
-  day of the month
etc...