tomato64/tomato64 #24: Issues with make

kuabi, 2024-01-04:
I'm have issues where rp-pppoe-tomato 3.15 isn't found.

> >>> rp-pppoe-tomato 3.15 Downloading
wget --passive-ftp -nd -t 3 -O '/home/les/development/tomato64/src/buildroot/output/build/.rp-pppoe-3.15.tar.gz.j2bRAb/output' 'https://dianne.skoll.ca/projects/rp-pppoe/download/OLD/rp-pppoe-3.15.tar.gz' 
--2024-01-04 22:58:54--  https://dianne.skoll.ca/projects/rp-pppoe/download/OLD/rp-pppoe-3.15.tar.gz
Resolving dianne.skoll.ca (dianne.skoll.ca)... 144.217.161.9, 2607:5300:201:3000::4c3d
Connecting to dianne.skoll.ca (dianne.skoll.ca)|144.217.161.9|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-01-04 22:58:54 ERROR 404: Not Found.

wget --passive-ftp -nd -t 3 -O '/home/les/development/tomato64/src/buildroot/output/build/.rp-pppoe-3.15.tar.gz.rIvCbY/output' 'https://sources.buildroot.net/rp-pppoe-tomato/rp-pppoe-3.15.tar.gz' 
--2024-01-04 22:58:54--  https://sources.buildroot.net/rp-pppoe-tomato/rp-pppoe-3.15.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.0.37, 104.26.1.37, 172.67.72.56, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.0.37|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-01-04 22:58:55 ERROR 404: Not Found.

wget --passive-ftp -nd -t 3 -O '/home/les/development/tomato64/src/buildroot/output/build/.rp-pppoe-3.15.tar.gz.F8B32r/output' 'https://sources.buildroot.net/rp-pppoe-3.15.tar.gz' 
--2024-01-04 22:58:55--  https://sources.buildroot.net/rp-pppoe-3.15.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.1.37, 172.67.72.56, 104.26.0.37, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.1.37|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-01-04 22:58:55 ERROR 404: Not Found.

closed 2024-01-05 by kuabi

lancethepants, 2024-01-06:
Now fixed.

lancethepants, 2024-01-06:
https://github.com/tomato64/tomato64/commit/22c9cfdb7c1c18767c8a925e77271d7092130bc0