tomato64/tomato64 #31: Mounting /opt partition problem

byniow, 2024-04-16:
I use nightly build 04-12-2024 and 04-15-2024. And /opt partition is not mounted. I use installer from iso image and answered yes about /opt partition question. I get:

```
unknown login: root
Password: 

Tomato64 x86_64 2024.1 
size: 27290 bytes (1021286 left)
 ======================================================== 
 Welcome to the x86_64 [Tomato64]
 Date: Tue Apr 16 2024
 Time: 10:47:41 up 0 min
 Load average: 0.44, 0.11, 0.03
 Mem usage: 1.8% (used 64.47 of 3603.40 MB)
 WAN : 10.1.1.51/24 @ 68:05:CA:xx:xx:xx
 LAN1: 192.168.1.1/24 @ DHCP:  - 
 ======================================================== 

root@unknown:/tmp/home/root# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root              1992552    767272   1104040  41% /
devtmpfs               1842600         0   1842600   0% /dev
tmpfs                  1844940       256   1844684   0% /tmp
/dev/sda2                90333      1010     82155   1% /nvram
root@unknown:/tmp/home/root# e2label /dev/sda4
opt
root@unknown:/tmp/home/root# mount -a
mount: /opt: can't find LABEL=opt.
root@unknown:/tmp/home/root# 
```

In dmesg nothing about /opt.

lancethepants, 2024-04-16:
The script probably still needs some tweaking.
What does the following return?
```
fdisk -l
blkid
```
If the opt partition wasn't successfully created the root filesystem should have expanded on reboot to fill the remaining space, but from your `df` statement it doesn't look like it has expanded.  Did you make sure to remove the installation media and boot from the installation disk? Booting from the ISO you will see 3 entries in grub, but booting from disk installation you will only see 2 entries.

byniow, 2024-04-16:
```
root@unknown:/tmp/home/root# fdisk -l
Disk /dev/sda: 14.91 GiB, 16013942784 bytes, 31277232 sectors
Disk model: 16GB SATA Flash 
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 4379A189-7DC1-493B-AAF9-27A2DD50BB8A

Device       Start      End  Sectors  Size Type
/dev/sda1       64    20543    20480   10M EFI System
/dev/sda2    20544   225343   204800  100M Linux filesystem
/dev/sda3   225344  4419647  4194304    2G Linux root (x86-64)
/dev/sda4  4421632 31277055 26855424 12.8G Linux filesystem
root@unknown:/tmp/home/root# blkid
/dev/sda2: LABEL="nvram" UUID="87446697-321d-4766-aa7f-e84778ddd0f5" BLOCK_SIZE="1024" TYPE="ext4" PARTLABEL="nvram" PARTUUID="fd0f8669-d2c0-43dd-a5bd-733c3183fbe6"
/dev/sda3: LABEL="rootfs" UUID="82a3229d-de13-4cc3-bf39-2c0928ae0cc8" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="root" PARTUUID="82a3229d-de13-4cc3-bf39-2c0928ae0cc8"
/dev/sda1: SEC_TYPE="msdos" LABEL_FATBOOT="boot" LABEL="boot" UUID="18BE-F6EC" BLOCK_SIZE="512" TYPE="vfat" PARTLABEL="boot" PARTUUID="4727a32c-f769-4ec3-ad0e-69c755e9bc8c"
root@unknown:/tmp/home/root# 
```

When i change in fstab LABEL=opt to /dev/sda4 and then mount works well, but after restart fstab is like before with LABEL entry.
Im sure to boot from SATA drive.

lancethepants, 2024-04-16:
Do you have another disk you can try? If you also manually format it and it doesn't stick I'm wondering if it's a disk issue.

lancethepants, 2024-04-16:
I make fstab point to LABEL=opt because in some scenarios it may exist on a different disk.  It's weird that the partition on your disk isn't remembering the partition label

byniow, 2024-04-16:
This disk work well on full linux distro. But ok, i replaced disk to traditional 1TB HDD and /opt partition mounted well.

lancethepants, 2024-04-16:
It may be possible Tomato64 isn't handling something right with the "16GB SATA Flash".   There is a couple things you could test for me if your willing.

First we could try completely wiping the disk "16GB SATA Flash"
```
dd if=/dev/zero of=/dev/sda bs=1M
```
Then install Tomato64 again

if that doesn't work, you could try this then

Reinstall Tomato64 to your "16GB SATA Flash" with opt partition.  Then run
```
mount /dev/sda3 /mnt
rm /mnt/usr/bin/expand_root_partition
```
I wonder if this could maybe be causing an issue, but I'm a little doubtful.

If that doesn't work, I'd be curious to see you wipe the disk, and manually create a new gpt partition, and create 4 partitions, format them with some labels with whatever fs, reboot the computer and see if all the labels still exist.

lancethepants, 2024-05-03:
As you mentioned to me now, same disk now in a different machine running T64 now works when creating /opt. Closing for now then.

closed 2024-05-03 by lancethepants