tomato64/tomato64 #32: Port forwarding not work

byniow, 2024-04-22:
[image]
I have setting like this. Working on Tomato on my Tenda AC15. On Tomato86 not work at all. Forwarding not work from outsine network and from inside (via nat loopback) 

Iptables -L
```
root@unknown:/tmp/home/root# iptables -L
Chain INPUT (policy DROP)
target     prot opt source               destination         
DROP       all  --  anywhere             anywhere             state INVALID
ACCEPT     all  --  anywhere             anywhere             state RELATED,ESTABLISHED
shlimit    tcp  --  anywhere             anywhere             tcp dpt:ssh state NEW
wwwlimit   tcp  --  anywhere             anywhere             tcp dpt:8443 state NEW
ACCEPT     all  --  anywhere             anywhere            
ACCEPT     all  --  anywhere             anywhere            
ACCEPT     icmp --  anywhere             anywhere             icmp echo-request state NEW,RELATED,ESTABLISHED limit: avg 3/sec burst 5
ACCEPT     icmp --  anywhere             anywhere             icmptype 30 state NEW,RELATED,ESTABLISHED limit: avg 3/sec burst 5
ACCEPT     udp  --  anywhere             anywhere             udp dpts:33434:33534 limit: avg 3/sec burst 5
ACCEPT     tcp  --  anywhere             anywhere             tcp dpt:8443
ACCEPT     tcp  --  anywhere             anywhere             tcp dpt:7682

Chain FORWARD (policy DROP)
target     prot opt source               destination         
           all  --  anywhere             anywhere            account: network/netmask: 10.1.0.0/255.255.254.0 name: lan 
ACCEPT     all  --  anywhere             anywhere            
DROP       all  --  anywhere             anywhere             state INVALID
monitor    all  --  anywhere             anywhere            
ACCEPT     all  --  anywhere             anywhere             state RELATED,ESTABLISHED
ACCEPT     esp  --  anywhere             anywhere            
ACCEPT     ah   --  anywhere             anywhere            
ACCEPT     udp  --  anywhere             anywhere             udp dpt:500
ACCEPT     udp  --  anywhere             anywhere             udp dpt:4500
wanin      all  --  anywhere             anywhere            
wanout     all  --  anywhere             anywhere            
ACCEPT     all  --  anywhere             anywhere            

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         

Chain monitor (1 references)
target     prot opt source               destination         
RETURN     tcp  --  anywhere             anywhere            WEBMON --max_domains 9999 --max_searches 9999 

Chain shlimit (1 references)
target     prot opt source               destination         
           all  --  anywhere             anywhere             recent: SET name: shlimit side: source mask: 255.255.255.255
DROP       all  --  anywhere             anywhere             recent: UPDATE seconds: 600 hit_count: 4 name: shlimit side: source mask: 255.255.255.255

Chain wanin (1 references)
target     prot opt source               destination         
ACCEPT     tcp  --  anywhere             10.1.1.3             tcp dpt:www
ACCEPT     tcp  --  anywhere             10.1.1.3             tcp dpt:https
ACCEPT     tcp  --  anywhere             10.1.1.6             tcp dpt:imap2
ACCEPT     tcp  --  anywhere             10.1.1.6             tcp dpt:587
ACCEPT     udp  --  anywhere             10.1.0.3             udp dpt:1194
ACCEPT     tcp  --  anywhere             10.1.1.6             tcp dpt:ssmtp

Chain wanout (1 references)
target     prot opt source               destination         

Chain wwwlimit (1 references)
target     prot opt source               destination         
           all  --  anywhere             anywhere             recent: SET name: www side: source mask: 255.255.255.255
DROP       all  --  anywhere             anywhere             recent: UPDATE seconds: 5 hit_count: 15 name: www side: source mask: 255.255.255.255
root@unknown:/tmp/home/root# 
```

lancethepants, 2024-04-22:
I haven't had any issues with port forwarding myself.  After disabling webmon report back if this is still an issue.

byniow, 2024-04-23:
I disabled webmon. Not helped. Reinstalled tomato with 04-18-2024 image. Not helped, routing still not work. Qos-> transfer rates not showed any transfers to 10.1.1.3 where i have nginx proxy. I try disable nat helpers, ndpi, etc, nothing helped. 
My hardware is a Igel H830 with Celeron J1900 with additional pci-e lan card, connection is a Fiber PPPoE on Vlan35 (via ONT).

lancethepants, 2024-04-23:
So I was trying to replicate your setup and was also having the same issue as you, until I realized my wireguard vpn I use to remote into my dev machine was also using the 10.1.0.0/23 range, so there were addressing conflicts. Once I figured that then I have no issue with port forwarding. I was wondering if it was an issue being a /23 network, but I'm not seeing that being an issue for me.  I'm now guessing there's some sort of conflict with your network setup.  Are you trying to setup the Tomato64 router behind your other router?

byniow, 2024-04-24:
Hi. This is config working on FreshTomato from years, the same network is on my replacement mikrotik works too. Tomato - when is the same address on more than one interface (propably this You called as address conflict) not start routing at all. Here routing works, but no port forwarding. Today i try install Tomato on Proxmox, maybye all problems gone.

lancethepants, 2024-04-24:
Update:
Out of band we figured out this is not a port forwarding issue. For some reason the OP's web server is extremely slow when serving up https content, but works fine under FreshTomato router and microtik. At the moment my thoughts are maybe an MTU issue? or could be something else. It's an interesting internet configuration, pppoe over vlan35.  Our troubleshooting was while he was running virtualized on proxmox, letting proxmox handle the vlan35 requirement for the connection, but he also experienced the same issue when running bare metal. It's a difficult scenario for me to duplicate, so if anyone with a similar internet setup (maybe another Polish ISP subscriber :) @shibby20 ) had a moment to test this out as well would be awesome.

shibby20, 2024-04-25:
PPPoE over vlan35 is very popular WAN configuration in Polish (Orange, Netia with ONT). Unfortunately my ISP is not one of them. 

One question to @byniow - do you have Tomato64 on baremetal on under Virtualization?

byniow, 2024-04-25:
Hi. First Tomato work on baremetal, now under Proxmox (easy restoring). But this behavior is very strange. I have mailcow with imap and smtp ports open. While in my phone i can receive mails via imap on port 143 i cannot send mialis via smtp on port 465. This ports was open. Similiar, on port 80 and 443 i have nginx proxy manager. Where HomeAssistant app work on 443 nextcloud not. The same port. From inside network via nat loopback behavios is identical. I replace tomato to mikrotik (switching cables phisically) immediately all start works as should without any problem.

byniow, 2024-04-26:
> One question to @byniow - do you have Tomato64 on baremetal on under Virtualization? Under proxmox i sugest you to Network configuration and switch from regular Network Device/Linux Bridge type to OVS Port/OVS Bridge.

I replaced hardware to i5 3gen and install on baremetal. Not helping with port forwarding.

lancethepants, 2024-04-29:
Should now be fixed with this commit https://github.com/tomato64/tomato64/commit/65eeabd27d1aebebe0bf628e90c0f66f7eb1c299

closed 2024-04-29 by lancethepants