tomato64/tomato64 #33: Tomato unexpectly hang

byniow, 2024-04-22:
I haveconfigured backups webmon, iptraffic  and bwmon ok /opt catalog. Like this:
[image]
And Tomato hang (no routing to internet but webgui works)
[image]
Still /opt is a problem?

lancethepants, 2024-04-22:
Web Monitoring currently has some issues, so it's probably best to disable it at the moment. I need to disable that feature until it gets fixed.

https://www.linksysinfo.org/index.php?threads/tomato64-development-discussion-thread.78306/page-2#post-348471

lancethepants, 2024-04-22:
87ba384562d3f654510834a96213c7ba2cee51f3 disables enabling Web Monitor until it's fixes. Hopefully resolves your other issue.

lancethepants, 2024-04-29:
Should now be fixed with this commit https://github.com/tomato64/tomato64/commit/87ba384562d3f654510834a96213c7ba2cee51f3
This commits prevents enabling webmon until I it gets fixed.  The kernel module needs some modernization yet. It doesn't seem to advance to the next entry when doing a save and fills up ram until things hang or crash.

closed 2024-04-29 by lancethepants