tomato64/tomato64 #34: New instalation and few issues

shibby20, 2024-04-24:
Fresh installation on Proxmox from ISO and setup-tomato command with /opt creation. 

1) i need to configure vlan2 and vlan5 (both tagged) so i decided to create similarly br2 and br5. But br1,br3,br4 was unsused. Gui allowed me to do this. Result: after reboot br2, br5, vlan2 and vlan5 interfaces was not bring up (ifconfig command and only br0 was up). When i changed bridges to be in order: vlan2->br1, vlan5->br2 (vlan0 and br0 was untouched) then all interfaces was up after reboot

[image]

[image]

[image]

[image]

2) I enabled remote access to GUI on port 4444 (HTTPS). Access worked correct. After reboot i cound not connect to GUI remotely (in iptables was all required entries). So i went to Admin Access page (locally), didnt change anything just click "save" and remote access alive. Another reboot and same thing, remote access doesn`t work, "save" on Admin Access page, remote access works again. I compared /etc/iptables files after reboot and after "save" click - both are the same... strange.

[image]

3) in /etc/iptables i saw opened port 7682. What is this?

[image]

nestat shows that this is ttyd process. Is this should be opened from WAN by default??

[image]

4) i configure second WAN with weight 0 (failover mode) - DHCP, manual DNS, Watchdog disabled for second WAN). Result:
- on overwie page shows Watchdog enabled...

[image]

That`s all at the moment :)

Best Regards

[image]

pedro0311, 2024-04-24:
ad4.: https://bitbucket.org/pedro311/freshtomato-arm/commits/0dc64507d7ee9f9bc8fffb08ed6cdb75caf4a699

pedro0311, 2024-04-24:
ad2.: Is
`iptables -nvL --line-numbers`
gives really the same with the same order?

pedro0311, 2024-04-24:
ad1: it was already discussed: bridge numbers must be added sequentially, otherwise - in some cases - it creates problems.

shibby20, 2024-04-24:
> ad1: it was already discussed: bridge numbers must be added sequentially, otherwise - in some cases - it creates problems.

ok, i get it, so GUI should not allowed me to select bridge number manually. 

BTW little Request: add field with description to bridge configuration. When we have few or lot bridges then remembering which bridge is for what is really difficurt

> ad2.: Is
> `iptables -nvL --line-numbers`
> gives really the same output with the same order?

yep
[image]

pedro0311, 2024-04-24:
> ok, i get it, so GUI should not allowed me to select bridge number manually.
>
>BTW little Request: add field with description to bridge configuration. When we have few or lot bridges then remembering which bridge is for what is really difficurt

OK, added to TODO list

>yep

`service firewall restart` instead of 'save' doesn't help?

shibby20, 2024-04-24:
> service firewall restart instead of 'save' doesn't help?

it doesn`t...

this is something with multiwan... When i change to single WAN then remote access works fine, but with 2 WANs in failover mode got an issue and some errors in console

[image]

Configuration is very simple

[image]

shibby20, 2024-04-24:
this is repeatable::
- erase configuration to default
- add vlan for wan2, reboot
- add wan2 with weight 0
- save

Result

[image]
 
 i checked also with weight 1 - result is the same.

pedro0311, 2024-04-24:
So maybe with this we can find differences?:
`echo ip route show:
ip route show
echo
echo ip addr show:
ip addr show
echo
echo ip rule show:
ip rule show
echo
echo ip neigh show:
ip neigh show
echo
echo tc -d qdisc show:
tc -d qdisc show
echo
echo ip netstat -rn:
netstat -rn
echo
echo route -n:
route -n`

lancethepants, 2024-04-24:
ad3: ttyd powers the new "System Commands" Page. Port 7682 is only opened on WAN when remote HTTPS is enabled.  I've set ttyd to always require login, so should be as secure as opening the web interface to WAN

shibby20, 2024-04-25:
> So maybe with this we can find differences?:

[tests.zip](https://github.com/tomato64/tomato64/files/15105394/tests.zip)

and as we say "a picture is worth a thousand words)", so i recorded a little video.

https://www.youtube.com/watch?v=uEhjUDKD2ww

Conversion to HD/4K is in progress, so it should be ready in 1 hour.

pedro0311, 2024-04-25:
@shibby20 : I'm rather confused with routing, so maybe you can help here?
How can I get a log like yours (in vmware) with errors when executing scripts?
I have nothing like that in my syslog...

HommeOursPorc, 2024-04-25:
Does the latest commit about remote access help in a MultiWAN scenario? 

https://bitbucket.org/pedro311/freshtomato-arm/commits/105436cc056e2e795530a9114c53d804d81828e4

pedro0311, 2024-04-27:
@HommeOursPorc : it's about routing not iptables.

pedro0311, 2024-05-14:
@shibby20 : build your image with "export TOMATO_EXPERIMENTAL=1", enable debug:

> nvram set interface_debug=1
> nvram set mwan_debug=1
> nvram commit

and watch syslog. It should help to find these issues.

lancethepants, 2024-05-14:
I added a new Debug Experimental option to the menu config, and fixed up the Makefiles to use it.

I've also compiled some images with Debug Experimental Enabled.
https://tomato64.org/files/x86_64/Experimental/

pedro0311, 2024-05-15:
OK, so this issue also didn't help:
https://bitbucket.org/pedro311/freshtomato-arm/commits/805e62e3c97e027cd3879b6cfbd318f0189472fa

This problem did not affect WAN such as DHCP or LTE, but there was a problem with others.

AdminMPTech, 2024-08-13:
Hi again, I was very busy for last few months. Today i fresh-installed Tomato64 Experimental, enabled debug in nvram, enabled remote access on port HTTPS 4444, configured 2nd wan as eth0.12 and configured it with weight 0 as failover.

After that i did:
- reboot router
- click save on basic networks page
- click save on admin access page
- download log

I added some comments into logs by "logger" command. Hope this will help to find an issue.

[syslog.txt](https://github.com/user-attachments/files/16595703/syslog.txt)

shibby20, 2024-08-13:
Hi again, I was very busy for last few months. Today i fresh-installed Tomato64 Experimental, enabled debug in nvram, enabled remote access on port HTTPS 4444, configured 2nd wan as eth0.12 and configured it with weight 0 as failover.

After that i did:

    reboot router
    click save on basic networks page
    click save on admin access page
    download log

I added some comments into logs by "logger" command. Hope this will help to find an issue.    
    
[syslog.txt](https://github.com/user-attachments/files/16595883/syslog.txt)

forzic, 2025-02-12:
Buonesera connessione pppoe utente e password e vlan id nel mio caso Tim 835 dove va inserito? che non mi sale la connessione?

scoobasteve83, 2026-01-08:
> ok, i get it, so GUI should not allowed me to select bridge number manually.
>
> BTW little Request: add field with description to bridge configuration. When we have few or lot bridges then remembering [...]

For VLANS groups same thing.

Indeed there should be a VLAN label instead of the Firmware ID Box as far as I am concerned that firmware ID box has zero to do with vlans and could be hidden period it just adds confusion for folks. For VLANS would be nice if this could fit....
I think it could if we used @wellloaded 's VLAN page and methodology it should give us a little room back.
A label for what the VLAN group is would almost make too much sense even in fresh tomato....

If we are cramped for space have some text boxes underneath the table that allows us to enter a label for each VLAN for reference. This is common in a lot of switch equipment for this reason. As it is right now I have a cheat sheet to remember stuff until I lose that then I gotta do duduction all over again.

scoobasteve83, 2026-01-08:
Is issue solved ?

lancethepants, 2026-09-11:
There's been lots of improvements to multiwan the last little while. Please re-open if you revisit and have any issues.

closed 2026-09-11 by lancethepants