tomato64/tomato64 #35: No access to GUI or SSH due to bad configuration settings

turnupthebassto11, 2024-06-16:
First attempt at VLANs, borked my config and can't use GUI or SSH but I do have a spare monitor and keyboard within reach. How can I roll back the settings or clear all setting back to default?

lancethepants, 2024-06-16:
Using the monitor and keyboard you can wipe the nvram with "rm /nvram/*" and then "reboot" will then restore to default settings.

lancethepants, 2024-06-24:
Were you able to reset things?

HommeOursPorc, 2024-06-24:
I don't want to hijack the thread, but I was in the same boat and I can confirm that rm /nvram/* effectively force a default restoration. 

About rolling back, I didn't try, can we use nvram save and nvram restore like for ARM FreshTomato?

lancethepants, 2024-06-24:
Tomato64 does have the ability to backup and restore nvram configurations in `Administration -> Configuration` like FreshTomato.

You cannot however use a FreshTomato backup on Tomato64 or vice versa. There is enough difference between the two that it would bork things

HommeOursPorc, 2024-06-24:
I meant the commands, on the CLI. I suppose copy/pasting what is in /nvram should get the job done if we do not have nvram save/restore commands.

lancethepants, 2024-06-24:
The nvram cli utility supports all the same functionality as FreshTomato as well.

turnupthebassto11, 2024-06-25:
I was, thank you for the follow up! Can this get added to the wiki for
other users?

lancethepants, 2024-06-27:
I've added a page [here](https://github.com/tomato64/tomato64/wiki/Factory-Reset)

closed 2024-06-27 by lancethepants