tomato64/tomato64 #36: Memory handling problems, Tomato64 on Proxmox crash when RAM is overfilled

HommeOursPorc, 2024-09-03:
There's a problem with Tomato64 on Proxmox regarding memory handling. When the RAM is overfilled Tomato64 crash, making it unreachable.

I didn't test with a bare metal setup, I don't know if the problem is with Proxmox or Tomato64.

I wonder if the use of a swap partition could solve or at least mitigate the problem. 

The problem can be easily replicated, just create a random file in the filesystem then copy it into /tmp until the RAM is maxed out. 

```
root@Tomato64-AP:/jffs# head -c 200m </dev/urandom >random.data
root@Tomato64-AP:/jffs# ls -la
total 204812
drwxr-xr-x    2 root     root          4096 Sep  3 08:52 .
drwxr-xr-x   27 root     root          4096 Jul 31 11:59 ..
-rw-r--r--    1 root     root     209715200 Sep  3 08:52 random.data
root@Tomato64-AP:/jffs# cat /proc/meminfo
MemTotal:         700112 kB
MemFree:          363092 kB
MemAvailable:     570112 kB
Buffers:            5392 kB
Cached:           255576 kB
SwapCached:            0 kB
Active:            30004 kB
Inactive:         239844 kB
Active(anon):        856 kB
Inactive(anon):    13120 kB
Active(file):      29148 kB
Inactive(file):   226724 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:          8896 kB
Mapped:            15912 kB
Shmem:              5096 kB
KReclaimable:      20460 kB
Slab:              34848 kB
SReclaimable:      20460 kB
SUnreclaim:        14388 kB
KernelStack:        1776 kB
PageTables:         1652 kB
SecPageTables:         0 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      350056 kB
Committed_AS:      21700 kB
VmallocTotal:   34359738367 kB
VmallocUsed:       13128 kB
VmallocChunk:          0 kB
Percpu:              536 kB
HugePages_Total:       0
HugePages_Free:        0
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
Hugetlb:               0 kB
DirectMap4k:       32664 kB
DirectMap2M:      747520 kB
root@Tomato64-AP:/jffs# cp random.data /tmp/r1
root@Tomato64-AP:/jffs# cat /proc/meminfo
MemTotal:         700112 kB
MemFree:          156292 kB
MemAvailable:     363544 kB
Buffers:            5392 kB
Cached:           460376 kB
SwapCached:            0 kB
Active:            30004 kB
Inactive:         444624 kB
Active(anon):        856 kB
Inactive(anon):   217900 kB
Active(file):      29148 kB
Inactive(file):   226724 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:          8896 kB
Mapped:            15912 kB
Shmem:            209896 kB
KReclaimable:      20924 kB
Slab:              35308 kB
SReclaimable:      20924 kB
SUnreclaim:        14384 kB
KernelStack:        1776 kB
PageTables:         1652 kB
SecPageTables:         0 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      350056 kB
Committed_AS:     226500 kB
VmallocTotal:   34359738367 kB
VmallocUsed:       13144 kB
VmallocChunk:          0 kB
Percpu:              536 kB
HugePages_Total:       0
HugePages_Free:        0
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
Hugetlb:               0 kB
DirectMap4k:       32664 kB
DirectMap2M:      747520 kB
root@Tomato64-AP:/jffs# cp random.data /tmp/r2
cp: write error: No space left on device
sh: out of memory
Connection to 192.168.1.64 closed.
~ $ ssh root@192.168.1.64
Connection closed by 192.168.1.64 port 22
~ $ ssh root@192.168.1.64
Connection closed by 192.168.1.64 port 22
```

HommeOursPorc, 2024-09-03:
Even the console is dead

[image]

lancethepants, 2025-11-30:
More ram is always good, or now we have zram feature available for all platforms.

closed 2025-11-30 by lancethepants