tomato64/tomato64 #38: Can't update using few latest nightly

HommeOursPorc, 2024-09-03:
The latest nightly won't update properly. I'm getting "error 2".

This one went through fine: 
tomato64-08.10.24-uefi-x86_64_v2+-update.tzst

Latest September 1 and 2 doesn't work. 
```
ERROR: Writing of tomato64-08.30.24-uefi-x86_64_v2+-update.tzst unsuccessful!!
ERROR: error code "2" returned during extraction
```

I didn't test them all but something is broken after the one from August 10. I don't know were is the turning point but I can confirm 08.10 went through just fine, 08.30, 09.01 and 09.02 errors out.

```
root@Tomato64-AP:/opt# t64up_016.sh https://tomato64.or
g/files/x86_64/Nightly/09-02-2024/tomato64-09.02.24-uef
i-x86_64_v2%2B-update.tzst
Connecting to tomato64.org (199.119.84.5:443)
saving to '/tmp/t64upwd/tomato64-09.02.24-uefi-x86_64_v2+-update.tzst'
tomato64-09.02.24-ue 100% |*******|  206M  0:00:00 ETA
'/tmp/t64upwd/tomato64-09.02.24-uefi-x86_64_v2+-update.tzst' saved
Verifying file... Done
Verifying checksum...
wget: server returned error: HTTP/1.1 404 Not Found
ERROR: Impossible to verify downloaded firmware checksum or checksum doesn't match!
Continue anyway? To continue type [yes], to cancel type [n]: yes

                POINT OF NO RETURN

Make sure you are using the right firmware before proceeding!

Firmware: tomato64-09.02.24-uefi-x86_64_v2+-update.tzst
System boot firmware: UEFI

Proceed dirty upgrade with firmware 'tomato64-09.02.24-uefi-x86_64_v2+-update.tzst' ?
To continue type [yes], to cancel type [n]: yes

Starting dirty upgrade with firmware 'tomato64-09.02.24-uefi-x86_64_v2+-update.tzst'
Stopping services... Done
Writing firmware tomato64-09.02.24-uefi-x86_64_v2+-update.tzst
Writing firmware... Done
ERROR: Writing of tomato64-09.02.24-uefi-x86_64_v2+-update.tzst unsuccessful!!
ERROR: error code "2" returned during extraction

Something went wrong. The router is NOT upgraded. The router will reboot.

To cancel the reboot and retry the upgrade manually press 'Ctrl+C'

^Cbooting in 2
Operation cancelled.
root@Tomato64-AP:/opt# t64up_016.sh https://tomato64.or
g/files/x86_64/Nightly/08-30-2024/tomato64-08.30.24-uef
i-x86_64_v2%2B-update.tzst
Connecting to tomato64.org (199.119.84.5:443)
saving to '/tmp/t64upwd/tomato64-08.30.24-uefi-x86_64_v2+-update.tzst'
tomato64-08.30.24-ue 100% |*******|  206M  0:00:00 ETA
'/tmp/t64upwd/tomato64-08.30.24-uefi-x86_64_v2+-update.tzst' saved
Verifying file... Done
Verifying checksum...
wget: server returned error: HTTP/1.1 404 Not Found
ERROR: Impossible to verify downloaded firmware checksum or checksum doesn't match!
Continue anyway? To continue type [yes], to cancel type [n]: yes

                POINT OF NO RETURN

Make sure you are using the right firmware before proceeding!

Firmware: tomato64-08.30.24-uefi-x86_64_v2+-update.tzst
System boot firmware: UEFI

Proceed dirty upgrade with firmware 'tomato64-08.30.24-uefi-x86_64_v2+-update.tzst' ?
To continue type [yes], to cancel type [n]: yes

Starting dirty upgrade with firmware 'tomato64-08.30.24-uefi-x86_64_v2+-update.tzst'
Stopping services... Done
Writing firmware tomato64-08.30.24-uefi-x86_64_v2+-update.tzst
Writing firmware... Done
ERROR: Writing of tomato64-08.30.24-uefi-x86_64_v2+-update.tzst unsuccessful!!
ERROR: error code "2" returned during extraction

Something went wrong. The router is NOT upgraded. The router will reboot.

To cancel the reboot and retry the upgrade manually press 'Ctrl+C'

^Cbooting in 6
Operation cancelled.
root@Tomato64-AP:/opt# free
              total        used        free      shared  buff/cache   available
Mem:         700840       41272      266504        5060      393064      584160
Swap:             0           0           0
root@Tomato64-AP:/opt# t64up_016.sh https://tomato64.or
g/files/x86_64/Nightly/08-10-2024/tomato64-08.10.24-uef
i-x86_64_v2%2B-update.tzst
Connecting to tomato64.org (199.119.84.5:443)
saving to '/tmp/t64upwd/tomato64-08.10.24-uefi-x86_64_v2+-update.tzst'
tomato64-08.10.24-ue 100% |*******|  204M  0:00:00 ETA
'/tmp/t64upwd/tomato64-08.10.24-uefi-x86_64_v2+-update.tzst' saved
Verifying file... Done
Verifying checksum...
wget: server returned error: HTTP/1.1 404 Not Found
ERROR: Impossible to verify downloaded firmware checksum or checksum doesn't match!
Continue anyway? To continue type [yes], to cancel type [n]: yes

                POINT OF NO RETURN

Make sure you are using the right firmware before proceeding!

Firmware: tomato64-08.10.24-uefi-x86_64_v2+-update.tzst
System boot firmware: UEFI

Proceed dirty upgrade with firmware 'tomato64-08.10.24-uefi-x86_64_v2+-update.tzst' ?
To continue type [yes], to cancel type [n]: yes

Starting dirty upgrade with firmware 'tomato64-08.10.24-uefi-x86_64_v2+-update.tzst'
Stopping services... Done
Writing firmware tomato64-08.10.24-uefi-x86_64_v2+-update.tzst
Writing firmware... Done
Successfully flashed tomato64-08.10.24-uefi-x86_64_v2+-update.tzst
Rebooting...

Read from remote host 192.168.1.64: Connection reset by peer
Connection to 192.168.1.64 closed.
```

HommeOursPorc, 2024-09-04:
I've setup a brand new VM using release 2024.3 with default configs and then tried to upgrade using the GUI with nightly 09.02.24 and it doesn't work.

[image]

lancethepants, 2024-09-05:
First do `rm -rf /run` and then it should update after that.

HommeOursPorc, 2024-09-05:
> First do `rm -rf /run` and then it should update after that.

This works, thanks. 

I'm wondering why it is necessary all of a sudden? Was this expected?

Should `rm -rf /run` be added to /sbin/upgrade ?

lancethepants, 2024-09-05:
Some applications compiled in Tomato64 put their pid in /run instead of /var/run.  It's possible that some people set the filesystem as read-only, and so /run as read-only stops xtables (iptables) from writing it's pid and stop the firewall from starting. So I changed /run to by a symlink to /var/run, but unfortunately messes up with doing an tzst update unless you first delete /run.

What's do you think is the best way to communicate this with the community? We can put 'rm -rf /run' in /sbin/upgrade, but they would need to perform an upgrade first, so kind of a chicken egg scenario.

HommeOursPorc, 2024-09-05:
Good question. Maybe a message in the download repo + a message on LI forums? Or on the project home page + your website home page?

This changed at which version? I tried going from nightly 09.02 to 09.04 and it worked just fine.

lancethepants, 2024-09-05:
Once you've done that `rm -rf /run` step once you're good going forward.

HommeOursPorc, 2024-09-05:
Cool 😎 

So, I think my issue have been answered.

I think you can close this once you found the perfect place to notify the community 🙂

closed 2025-04-06 by lancethepants