tomato64/tomato64 #40: vlan + pppoe

lancethepants, 2025-02-12:
> Good evening pppoe connection user and password and vlan id in my case Tim 835 where should it be entered? why isn't my connection working? 

 _Originally posted by @forzic in [#34](https://github.com/tomato64/tomato64/issues/34#issuecomment-2654432489)_

forzic, 2025-02-12:
thanks now it works but glinet mt6000 does not go to 2500gb my internet is 2500/1000 but in download only 900mb what can I do? is there something in the firewall to enable?

lancethepants, 2025-02-12:
Oh wow, you have the fastest internet of any gl-mt6000 user using Tomato64 that I've heard of so far.  I actually still need to look into what it will take to enable hardware acceleration for this router on Tomato64.  As of yet it has not been implemented in T64. But now that I know someone has multi-gig speeds and is wanting to run T64, I will make this a priority.  

Through some experimentation someone in the forums discovered that if you disable the 2.5g lan port (by unchecking it in advanced-vlan) that you can get a full 1G down by using just the 1G lan ports while using pppoe.   It seems to be an issue specific to pppoe users that they cannot use both 2.5G port and get over 700MB on T64 or OpenWrt **IF** they do not enable hardware acceleration.  DHCP connections for example don't have this issue.  So that's one way to get a little bit better download speed with T64 on the GL-MT6000 at the moment, but I will look into hardware acceleration now.

forzic, 2025-02-12:
I will wait for developments I hope in version 7 of tomato64 which implements hardware acceleration

forzic, 2025-02-12:
It's a real shame not to be able to use mt6000 with WAN at 2500

forzic, 2025-02-13:
[image]

lancethepants, 2025-02-13:
I can't say how long it will take me for sure.
https://www.linksysinfo.org/index.php?threads/new-tomato64-port-gl-inet-flint2-gl-mt6000.78829/#post-353269

OpenWrt's and Tomato64's firewall methods are very different, so it won't be as easy as copy and paste from them.

lancethepants, 2025-02-27:
I've got something for you to test, Software and Hardware Routing/NAT offloading
https://tomato64.org/files/GL-MT6000/Nightly/02-26-2025/

Setting found in Advanced -> Firewall
[image]

lancethepants, 2025-03-06:
@forzic 

2025.1 Release Candidate now available, which includes Routing/NAT Offload support.
I'd love to see how it performs with your 2.5G internet if you could give it a test.

https://tomato64.org/files/GL-MT6000/Releases/2025/2025.1-rc1/

forzic, 2025-03-06:
Thank you so much for what you do tonight I will test the new release and let you know in any case I wanted to ask you if tomato supports load balance and feilover and how to set it up thanks

lancethepants, 2025-03-07:
> in any case I wanted to ask you if tomato supports load balance and feilover and how to set it up thanks

Yes... but from reports I've heard it's got some issues.  I think the same is true for FreshTomato as well.  
I have wanted to check out mwan3 from OpenWrt and see if it's something that could be implemented in Tomato.

lancethepants, 2025-03-07:
You would first go to Advanced -> VLAN, and assign a second WAN port as WAN1 bridge. Then go to Basic -> Network, and setup each of your wan connections and give them a weight.

forzic, 2025-03-07:
anyway now the 2.5gb line is working very well full bandwidth

forzic, 2025-03-07:
you should somehow implement pocket stering to divide the load on each cpu core

forzic, 2025-03-07:
what kernel is installed on now?

forzic, 2025-03-07:
currently kernel 6.6.80 is out>>>>could you help me remotely with TeamViewer?

forzic, 2025-03-07:
[image]

[image]

lancethepants, 2025-03-07:
For the time

> currently kernel 6.6.80 is out>>>>could you help me remotely with TeamViewer?

Currently I just plan to update the OpenWrt components when they make a point release, e.g. 24.10.1
Are you looking for help for multiwan?

<br>

>could take inspiration from the build of the developer pesa1234

I forgot about the pesa build.  I'll keep it in mind. He's got some good features seemingly not found in mainline OpenWrt
Packet steering is in mainline isn't it?

<br>

> are all drives MTK? or open source?

I'm using the open source drivers as found in mainline OpenWrt, mt76

forzic, 2025-03-08:
yes I would need help with multiwan and load balance on LAN 4 of the mt6000 with Tomato64,with TeamViewer this afternoon is it possible? where are you from where do you live because maybe there is a time zone I am in Italy...

forzic, 2025-03-08:
Pocket steering la curato pesa1234 su snapshot controlla suo rapisory

jamesskala, 2025-03-08:
I have the 2025 rc1 loaded on my GL-GT6000.  Working great.  
My Current Configuration is
1. Multi Wan (Load Balancing) 1:1 Ratio COMCAST Cable and Kinetic FTTH
2. Two Lan Networks
3.  Wifi working great

I'm currently getting my 1 Gig on LAN and Wi-Fi no issues (HW Offloading is On)

Very excited where this is going.  I have 2 of these routers.  Looking to the Flint 3 also.

lancethepants, 2025-03-09:
@jamesskala 
Thanks for the input and feedback!  I'm glad that it's working well for you.

@forzic 
My apologies, I do live in a much different time zone (UTC -7) and had a busy weekend.
Let me see if I can help with a couple of screenshots first go get you going.
<br>
So basically pick which port you would like to be your secondary Wan port.  
Remove that port from any other network and assign it to WAN1 Bridge
[image]

<br><br>
Then on the Basic -> Network page under "Number of logical WANs" you'll be able to select "2WAN"
Then below you will have WAN0 Settings and now WAN1 Settings where you can then connection type.

Check out this link and read the sections about the weight.
https://wiki.freshtomato.org/doku.php/basic-network

[image]

forzic, 2025-03-09:
for future updates do I upgrade from web ui with .tzst file? why do I already see nightly version but .bin version??? where do I get the right version?

forzic, 2025-03-09:
[image]

lancethepants, 2025-03-10:
Sorry, currently the only supported way to update the build is from the uboot environment. I still need to implement updating through the Web UI.

The status overview does show cpu and memory usage.  In the future I will show cpu and radio temperatures and disk usage as well.

forzic, 2025-03-10:
hello default username and password Tomato64?

lancethepants, 2025-03-10:
Yes, for wifi username/password is Tomato64/glmt6000

forzic, 2025-03-10:
no I meant access to the router 192.168.1.1 username and password

lancethepants, 2025-03-10:
> no I meant access to the router 192.168.1.1 username and password

root/admin

lancethepants, 2025-04-15:
It's now possible to update through the Web UI.

https://www.linksysinfo.org/index.php?threads/new-tomato64-port-gl-inet-flint2-gl-mt6000.78829/page-17#post-358226
https://tomato64.org/files/GL-MT6000/Nightly/04-14-2025/

If you're jumping from an older release, update from the Web UI two times.  The first time it will probably mention kernel error. The 2nd time it should update successfully without error.

lancethepants, 2025-04-28:
@forzic 
Packet Steering has now been implemented in the UI.

https://github.com/tomato64/tomato64/commit/ee09a3e2b85dc6b52f896e3223520215aa1a3836
https://www.linksysinfo.org/index.php?threads/new-tomato64-port-gl-inet-flint2-gl-mt6000.78829/page-22#post-358649

forzic, 2025-06-07:
thank you so much for the great job done number one

lancethepants, 2026-01-16:
> Is there no way to integrate kernel 6.12?

Kernel 6.12 is now available to test in the latest nightly builds
https://www.linksysinfo.org/index.php?threads/tomato64-development-discussion-thread.78306/page-22#post-365292

It does seem like everything so far has been addressed. If no more issues I will close say in a week. Else feel free to open another or post in the forums!